google/makerspace-auth #9: Board: Make the hardware cheaper

thatch, 2017-05-23:
I've been kicking around an idea for an ESP8266 board with fewer buttons.

ParthenonTech, 2018-08-07:
Would love that.  I'm wondering if this version or an upcoming one might work on a Pi Zero, too.

thatch, 2018-08-10:
@ParthenonTech This version should work just fine on a Pi Zero (although the 3rd mounting hole will be in open air, so you might want a 3d printed bracket).  I don't think it would shrink well down to that side without removing some features (and the enclosure still pretty much dominates the cost).  Do you have more info on your application, about what tool and mounting you want?

ParthenonTech, 2018-08-10:
I wasn't thinking about the size, just the BOM cost.  Anything to make it a no-brainer to attach one to just about anything in the makerspace that requires training.