google/makerspace-auth #11: C2 capacitor sizing

brianredbeard, 2018-03-01:
The C2 filter capacitor is specified by part number [C3216JB1E476M160AC][https://www.digikey.com/product-detail/en/tdk-corporation/C3216JB1E476M160AC/445-11717-1-ND/3953383] is flagged by the manfucaturer "Not for use in new designs".

For a ceramic capacitor of that high of a voltage/capacitance it seems that its a pretty niche part.

Sticking in a 1206 footprint it appears that one can go as high as 33µF when sticking to 12v or 47µF if the voltage can drop to 10v.

Are there any suggestions for replacement parts?

thatch, 2018-03-01:
My last build used EMK316BBJ476ML-T but that's out of stock at both digikey and mouser now too.

Originally, I was intending to PWM the LEDs, and included C2 defensively to cope with switching currents on underpowered power supplies... but driving with a nice power supply and slow output changes empirically works fine without C2.

Happy to answer other questions as well.

brianredbeard, 2018-05-20:
Understood.  I think this documentation sufficient for the time being to close the issue out.

closed 2018-05-20 by brianredbeard