google/makerspace-auth #22: Inquiry: Hookup of additional devices

brianredbeard, 2018-05-23:
I''m currently thinking through a personal need as well trying to be a bit more forward thinking.

Originally, I was going to make the suggestion of exposing UART via a 4 pin connector.

Then I realized that the pins for SPI0, I2C, _and_ UART are all unpopulated.  Is there an opinion on whether the strategy is to stack boards, use a pass through header, or potentially expose any of these pins via a header in the future?

thatch, 2018-05-23:
Avoiding those pins was an intentional design choice.  We've had spaces asking about displays in particular since the beginning.

For prototyping, I've tacked bodge wires on the UART pins and I think a stacking header on an intermediate hat would work fine.

Longer term, if you can find the board space to add those headers, go for it and send a PR.  The easiest place is probably where the URL is right now, and expand the board a little.  You could probably also sneak in the serial header where the 5V trace is near J1.

thvortex, 2018-05-23:
Just FYI re: UART.

My understanding is that the default device tree on a Raspberry Pi 3/ Zero
W connects the crappier "mini UART" to those external pins and it uses the
good PL011 UART for Bluetooth (Serial Port Profile?).

Probably the main side effect is that CPU frequency scaling is disabled
when using those external UART pins in the default config. You can apply a
device tree overlay to switch the UART assignment or to disable Bluetooth
outright.

See: https://www.raspberrypi.org/documentation/configuration/uart.md