google/makerspace-auth #55: Modifying to control a power strip via 3v-60v DC signal

bklebke, 2025-01-15:
Hello! Hoping to use this in SFO-SPE to control something like this: https://abra-electronics.com/electromechanical/relays/miscellaneous-en-4/2935-ada-controllable-four-outlet-power-relay-module-version-2-power-switch-tail-alternative.html

In my exploration, using J12/J13 doesn't seem to work -- I'm guessing because these are one of the two voltage-free switches mentioned in /hardware's readme. If I'm understanding the specs of that strip right (i.e., it needs a signal high/low to modulate the power strip), I'd need to hook into the high-current relay instead and modulate that. 

Any tips as to which plug that is  and how to address it? Am not an EE by training, so inspecting the trace didn't actually do much for me. 

@thatch for viz -- saw you helped put this together originally and are recently active on GH, so would be super appreciative of any help here. Thanks for your work on this to date, and hope all's well!

thatch, 2025-01-15:
The 3v listed voltage makes me wonder if they are avoiding the diode drop by doing something slightly more clever; I don't think it would be a mosfet as ideal diode due to the high end voltage listed; perhaps it's a circuit like:

```
IN+ --- R ---+----------+
             |          |
         [rev zener]    R
             |        [opto led]
             |          |
IN- ---------+----------+
```

then I'd expect around 30mA, still suitable for any of those listed headers

bklebke, 2025-01-15:
Thank you - deeply appreciate the quick and thoughtful response! 

It looks like we're using J1, J2, and J4 right now - for on/offbuttons and a buzzer, respectively

J7, J3/5, and J8 are all free. From http://www.digital-loggers.com/iotfaqs.html they claim 0.2ma to trigger, which seems low - but all are able to do this. 

Assuming I move the connectors to pins 1/3 on J7, is the config update as simple as modifying 

enable_output = Relay:ActiveHigh:29, Relay:ActiveHigh:31

to 

enable_output = Relay:ActiveHigh:40 
?

I know it's technically not a relay on J7, but we'd need a constant signal so I'm not sure the button or buzzer types are appropriate.

Thanks again!

bklebke, 2025-01-15:
Fantastic, was able to get it up and running with your help. Will add documentation for this going forward in case others use a similar approach. Really really appreciate your help!

closed 2025-01-15 by bklebke